Duet3D/ConfigTool #1: Z Probe: Switch / add explicit option for inductive probes NO & NC

Kriechi, 2018-11-10:
IMO inductive probes a quite common - it would be nice to directly list them so people don't have to think about what to pick instead (switch? unmodulated?). Similar as you already have it for endstops (active high, NO, NC, ...)

Also, does it make sense to have a trigger value for simple switches? Sure, RRF needs to be configured with something (or does it?)

chrishamm, 2018-11-12:
So are you talking about these two probe types? https://duet3d.dozuki.com/Wiki/Connecting_a_Z_probe#Section_NPN_output_normally_open_inductive_or_capacitive_sensor I guess I could add support for them although you are actually the first one who requests this :-)

The Z-probe measurment that is compared with the trigger threshold is usually filtered, so the value can actually make a difference depending on the selected probe speed. I guess 500 is a good starting value though. Besides it's been part of the first-generation config tool and it is still used in the template of the second version, so I'd rather leave it in there.

Kriechi, 2018-11-12:
yes - I'm using NPN inductive probes and they only give me a digital input signal - high or low. No need to use the analog measurement here.

Thinking about it some more, there is also the next section "Endstop configuration", which again has Z config options... not sure if or how this should be combined: Z-probes vs. Z-endstop.

My point was simply, that it wasn't obvious to me which config I have to choose.
In my manually written config files, I am using:
`M558 P4 H3 F400 T7000     ; Use E0_STOP as inductive-NPN probe, fast` and
`M558 P4 H3 F50 T7000 A3   ; Use E0_STOP as inductive-NPN probe, slow, multi-probe`
in my macros and homing files.

cleverer, 2020-01-08:
Is this still up for debate? I would be interested in such a functionality, as I'm not that confident with those config files and what I have to change where manually…