Duet3D/ConfigTool #15: Endstops:Z-Probe:BLTouch

Stephan-Diemer, 2019-11-18:
If a PWM Channel on the Duex X is chosen for the BLTouch servo channel it will still write the last chosen heater channel to the config.g deployprobe.g and retractprobe.g including the invert signal option

example:
config.g
...
M307 H3 A-1 C-1 D-1 instead of M307 H7 A-1 C-1 D-1
....

deployprobe.g
...
M280 P3 S10 I1 ; deploy BLTouch instead of M280 P7 S10 ; deploy BLTouch
...

chrishamm, 2019-11-23:
If I select BLTouch with RRF2, Duex5 and PWM on E6, it generates

M307 H7 A-1 C-1 D-1                     ; disable heater on PWM channel for BLTouch

in config.g and

M280 P7 S10 I1 ; deploy BLTouch

in deployprobe.g. So what's wrong here?

Stephan-Diemer, 2019-11-24:
I don't think it happens with the Duet but with the Duex ports. Try PWM1 to PWM5 on the expansion board itself. Still doesn't work for me.

chrishamm, 2019-12-09:
Thanks, I could reproduce this and will fix it in the next version.

chrishamm, 2019-12-09:
Fixed in v2.1.4.

closed 2019-12-09 by chrishamm