Duet3D/ConfigTool #16: Hasownproperty

fish850, 2019-11-20:
I have an issue trying to use an existing json file.

I get the following error:

Error: The specified file could not be read!
Type Error: Cannot read property 'hasOwnProperty' of null

Here is my json file:

{"board":"duetwifi10","expansion_boards":[],"firmware":2.03,"standalone":true,"nvram":false,"auto_save":{"enabled":false,"save_threshold":10,"resume_threshold":11,"gcodes_to_run":"M913 X0 Y0 G91 M83 G1 Z3 E-5 F1000"},"display":{"type":0,"encoder_steps":4,"spi_frequency":2000000,"menus":[{"name":"main","value":""}],"images":[]},"geometry":{"type":"cartesian","mins":[0,0,0],"maxes":[250,290,200],"delta_radius":105.6,"homed_height":250,"low_dive_height":false,"max_carriage_travel":260,"print_radius":85,"rod_length":215,"z_min":0},"drives":[{"direction":1,"microstepping":16,"microstepping_interpolation":true,"steps_per_mm":100,"instant_dv":15,"max_speed":200,"acceleration":1000,"current":800,"driver":0,"driver_v3":"0.0","endstop_pin":"xstop","endstop_type":2,"endstop_location":1},{"direction":1,"microstepping":16,"microstepping_interpolation":true,"steps_per_mm":100,"instant_dv":15,"max_speed":200,"acceleration":1000,"current":800,"driver":2,"driver_v3":"0.1","endstop_pin":"ystop","endstop_type":2,"endstop_location":2},{"direction":1,"microstepping":16,"microstepping_interpolation":true,"steps_per_mm":1080,"instant_dv":0.2,"max_speed":3,"acceleration":20,"current":800,"driver":1,"driver_v3":"0.2","endstop_pin":null,"endstop":null,"endstop_type":2,"endstop_location":1},{"direction":1,"microstepping":16,"microstepping_interpolation":true,"steps_per_mm":92.59,"instant_dv":2,"max_speed":50,"acceleration":250,"current":1600,"driver":3,"driver_v3":"0.3"},{"direction":1,"microstepping":16,"microstepping_interpolation":true,"steps_per_mm":92.59,"instant_dv":2,"max_speed":50,"acceleration":250,"current":1600,"driver":4,"driver_v3":"0.3"}],"idle":{"used":true,"factor":50,"timeout":30},"homing_speed_fast":30,"homing_speed_slow":6,"travel_speed":100,"z_dive_height":5,"slow_homing":false,"probe":{"type":"noprobe","recovery_time":0.4,"trigger_height":2.5,"trigger_value":500,"x_offset":0,"y_offset":0,"speed":2,"deploy":false,"points":[],"pwm_channel":3,"pwm_inverted":true,"pwm_pin":null,"input_pin":"zprobe.in","modulation_pin":"zprobe.mod"},"bed_is_nozzle":false,"bed":{"present":false,"use_pid":false,"heater":0},"chamber":{"present":false,"use_pid":false,"heater":2},"heaters":[null,{"temp_limit":280,"scale_factor":100,"series":4700,"thermistor":100000,"beta":4785,"a":0.0005717725,"b":0.0002416626,"c":1.300767e-7,"channel":1,"sensor":1,"output_pin":"e0heat","sensor_type":"thermistor","sensor_pin":"e0temp"},{"temp_limit":280,"scale_factor":100,"series":4700,"thermistor":100000,"beta":4785,"a":0.0005717725,"b":0.0002416626,"c":1.300767e-7,"channel":2,"sensor":1,"output_pin":null,"sensor_type":null,"sensor_pin":null}],"num_nozzles":2,"toolchange_wait_for_temperatures":true,"generate_t_code":false,"tools":[{"mix_ratio":[],"number":0,"name":"","extruders":[0],"heaters":[1],"fans":[],"x_offset":0,"y_offset":0,"z_offset":0},{"mix_ratio":[],"number":1,"name":"","extruders":[1],"heaters":[2],"fans":[],"x_offset":0,"y_offset":0,"z_offset":0}],"compensation_x_offset":15,"compensation_y_offset":15,"peripheral_points":3,"halfway_points":3,"calibration_factors":6,"probe_radius":85,"mesh":{"x_min":15,"x_max":180,"y_min":15,"y_max":180,"radius":85,"spacing":20},"home_first":false,"orthogonal":{"compensation":false,"height":50,"deviations":[0,0,0]},"network":{"enabled":true,"mac_address":"","name":"Zombie CubeX","password":"","ssid":"","ssid_password":"","dhcp":true,"ip":"192.168.1.20","netmask":"255.255.255.0","gateway":"192.168.1.254","protocols":{"http":true,"ftp":false,"telnet":false}},"fans":[{"name":"","value":100,"inverted":false,"frequency":500,"thermostatic":false,"heaters":[1,2],"trigger_temperature":45,"output_pin":"fan0"},{"name":"","value":100,"inverted":false,"frequency":500,"thermostatic":true,"heaters":[2,1],"trigger_temperature":45,"output_pin":"fan1"},{"name":"","value":100,"inverted":false,"frequency":500,"thermostatic":true,"heaters":[0,1,2],"trigger_temperature":45,"output_pin":null}],"custom_settings":""}

fish850, 2019-11-21:
I believe this may have been caused by not having a heated bed set up yet.

chrishamm, 2019-11-23:
Yes, that's right - it's been fixed in v2.1.3. Thanks for your report!

closed 2019-11-23 by chrishamm