Duet3D/ConfigTool #17: Heaters: Duet with FW 3.x don't support 3 in 1 hotend

kabroxiko, 2019-11-29:
I have a 3 in 1 hotend that have 3 heaters and 1 sensor
which means, I must configure more than one sensor on the same sensor.

https://www.aliexpress.com/item/32840733655.html

chrishamm, 2019-12-09:
It looks like the mix ratio inputs are broken - this will be fixed soon. Is that what you were trying to explain? Everything else seems to be working well. You need to define three extruders and one sensor assigned to a single tool to get a 3 in 1 hotend working.

kabroxiko, 2019-12-15:
I can't do that. Because when you add the second extruder, the e0temp sensor become grayed (disabled)

chrishamm, 2019-12-15:
A heater is not an extruder. Add three extruders E0, E1, and E2, and map the nozzle heater to e0temp. Then create a tool with all three E drives and a single nozzle heater mapped to it.

kabroxiko, 2019-12-17:
I know that a heater is not an extruder. Check the product link on my first post, my printer has 3 extruders (E0, E1 and E2) and 1 nozzle in 1 hotend with 1 sensor (e0temp) and 3 heaters connected (e0heat, e1heat and duex.e2heat) wi¡hich means it heat faster than with 1 heater.

chrishamm, 2020-11-16:
RepRapFirmware does not support multiple ports for a single heater yet. Hence we suggest to connect them to a single wire and to map them all to a single port. That shouldn't be a problem because they're only 20W per cartridge.

closed 2020-11-16 by chrishamm