Duet3D/ConfigTool #19: Incorrect homing scripts: H2 X and Y move

droftarts, 2019-12-19:
Ref: https://forum.duet3d.com/topic/13409/large-core-homing-issue/6

CoreXY setup, RRF 2.03-2.05 and RRF 3 setting, Duet WiFi, RRF Config 2.1.4 produces homeall.g and homez.g with G1 H2 move following homing of X and Y:
```
G90                     ; absolute positioning
G1 H2 X15 Y15 F6000     ; go to first bed probe point and home Z
G30                     ; home Z by probing the bed
```
On CoreXY, H2 pure move of X and Y positively cause no Y move, only X. Remove H2, it's not needed.

Ian

Dud3z, 2019-12-28:
same here:
https://forum.duet3d.com/topic/13525/problems-with-auto-bed-leveling-and-bltouch/3?_=1577527213237

chrishamm, 2020-01-07:
Thanks, fixed in v2.1.5.

closed 2020-01-07 by chrishamm