Duet3D/ConfigTool #24: Substitute endstop types "Active low" and "Active high" with "Switch"

chrishamm, 2020-01-15:
Substitute endstop types "Active low" and "Active high" with "Switch" - or - remove the invert options from the endstop pin list on the IO Mapping page. That's because M574 S0 is no longer supported

Pumlux, 2020-03-18:
I just run into this issue and it take some time to find a manual workaround. 
Here the findings I made, it might be useful for others running into the same issue.

actual status :
The config generator allows to select switch type active high (NC) but this seems to be no longer supported by RRF 3.0

seen behavior :
The Duet is not showing an error message during boot time (but if you send the M574 command manually). When running the homing command, the endstop is interpreted as always triggered

result : 
endstop signal is wrong and homing is getting wrong positions

manual workaround :
correct relevant config.g lines like this example for x-axis
M574 X1 S0 P"xstop"   ; not working command from config tool
M574 X1 S1 P"!xstop"  ; corrected command, note you need S1 AND invert the pin signal by using the '!' before the pinsignal name

chrishamm, 2020-05-27:
Will be fixed in the upcoming CT 3.1.0.

closed 2020-05-27 by chrishamm