Duet3D/ConfigTool #32: Issue setting config with duet 2 wifi and 3.0 and later

Mike-Hawkins, 2020-05-28:
Failed to generate file config.g:

ReferenceError: ejs:21
    19| <%		}
    20| 	} else { -%>
 >> 21| M669 K<%- sParam %> ; select CoreXY mode
    22| <%	}
    23| 	if (template.display.type) { -%>
    24| M918 P<%- template.display.type %> E<%- template.display.encoder_steps + ((template.firmware >= 2.03) ? ` F${template.display.spi_frequency}` : '') %> ; configure direct-connect display

sParam is not defined

DeadNewbie, 2020-05-28:
Mike-Hawkins, I was just doing the same, the error go's away when you select any other printer type than Cartesian.

Not sure if the resulting files are correct though or what you have to change to put it back to cartesian.

DeadNewbie, 2020-05-28:
Looks like there is an extra } on line 19 breaking the logic in the if conditional statements.

I dont know the .ejs code environment so lets wait for Chris to look at it.

Mike-Hawkins, 2020-05-28:
Yea I don't know ejs either but, I took used the corexy files and figured I should post the issue.

jmar007, 2020-05-28:
I'm seeing an error now as well. It worked the other day but now when I try to configure my Cartesian Anet A8 for my new duet2 wifi I get the error below.

Failed to generate ZIP bundle:

Failed to create config.g: ReferenceError: ejs:21
    19| <%		}
    20| 	} else { -%>
 >> 21| M669 K<%- sParam %> ; select CoreXY mode
    22| <%	}
    23| 	if (template.display.type) { -%>
    24| M918 P<%- template.display.type %> E<%- template.display.encoder_steps + ((template.firmware >= 2.03) ? ` F${template.display.spi_frequency}` : '') %> ; configure direct-connect display

sParam is not defined

chrishamm, 2020-05-28:
Thanks for the report, it's fixed now.

closed 2020-05-28 by chrishamm