Duet3D/ConfigTool #33: CoreXY Sensorless Homing

eseglem, 2020-05-29:
The current tool does not generate home*.g scripts that are useful for sensorless homing of a CoreXY printer (and probably any printer, but I can't test that at this time). 

Example working homex.g
```
M400                          ; wait til stuff stops
M913 X30 Y30                  ; drop motor currents to 30% 

M201 X1000 Y1000              ; reduce acceleration on X/Y to stop false triggers
M915 X Y H7 R0 F0 H400        ; both motors because corexy; Sensitivity 7, don’t take action, don’t filter, 400steps/sec

G91                           ; set relative
G1 Z5 F1200 H2
G1 H1 X-300 F4000             ; move left 300mm, stopping at the endstop
G1 X10 F1200                  ; move away from end
G1 Z-5 F1200 H2

G90                           ; back to absolute positioning
M400                          ; wait again
M913 X100 Y100                ; motor currents back to 100%
M201 X3500 Y3500              ; accel back to original
```

Y is basically the same but using Y-<Max> on the H1 instead of X-<Max>. 

I understand it might be hard to template out all possibilities, but would it be possible to add a toggle to not generate them in the zip, and override what I have made when updating something. Or potentially allow for storing of custom scripts (or even templates?) in the json file so I can paste them into a box, like the custom settings at the end and have them correct every time? 

It would also be useful to add the M915 to the `config.g` when sensorless is picked, but that's easy enough in the custom config part if you don't want to have to support it. The note with the wiki link is totally reasonable.

chrishamm, 2020-06-09:
The config tool explicitly says that sensorless homing requires extra steps. I know changing the acceleration and motor currents is needed for some machine configurations but not for all - in fact I have a machine that does not require any changes to the accelerations, currents, and StallGuard parameters.

M915 can be put into config.g, there is no reason to put it into the homing macros.

I'll change the backend of the config tool in foreseeable future to enable greater flexibility. Once that change is complete, I might add more SG settings.

chrishamm, 2025-01-07:
Those changes have been implemented as part of ConfigTool 3.5+.

closed 2025-01-07 by chrishamm