Duet3D/ConfigTool #38: PWM Control Channel (BLTouch only)

hinotori74, 2020-06-21:
Hi all,

With Probe BLTouch is not posible select **"exp.heaterX"** (X=3,4,5,6 or 7) on **"PWM Control Channel (BLTouch only)"**, only permit select duetx.pwmX (X=1,2,3,4 or 5).

Thanks

chrishamm, 2020-06-24:
It is possible but the pin names are different:
```
	{ PortCPin(3),	PinCapability::wpwm,	"exp.heater3,exp.8,!duex.e2heat,!duex.pwm1" },
	{ PortCPin(5),	PinCapability::wpwm,	"exp.heater4,exp.13,!duex.e3heat,!duex.pwm2" },
	{ PortCPin(8),	PinCapability::wpwm,	"exp.heater5,exp.18,!duex.e4heat,!duex.pwm3" },
	{ PortCPin(11),	PinCapability::wpwm,	"exp.heater6,exp.23,!duex.e5heat,!duex.pwm4" },
	{ PortAPin(15),	PinCapability::wpwm,	"exp.heater7,exp.31,!duex.e6heat,!duex.pwm5" },
```
So exp.heater3 corresponds to duex.pwm1 etc.

closed 2020-06-24 by chrishamm

hinotori74, 2020-06-24:
I had tried putting **"duex.pwm1"** but it didn't work. According to the above it should be **"duex.pwm1"** or **"! duex.pwm1"**. The website (https://configtool.reprapfirmware.org) leaves it as ** "duex.pwm1" **.

hinotori74, 2020-06-24:
With **"duex.pwm1"** the bltouch pin was never activated. With **"exp.heater3"** it works perfectly.

hinotori74, 2020-06-24:
I confirm that **"!duex.pwm1"** works correctly. The website (https://configtool.reprapfirmware.org) leaves it as **"duex.pwm1"**, the **"!"** sign is missing

jhbruhn, 2020-07-04:
Using duex.pwm1 in the ConfigTool for what was previously exp.heater3 does not work for me, the BL Touch Probe is never deployed.

natthapolvanasrivilai, 2020-07-05:
> Using duex.pwm1 in the ConfigTool for what was previously exp.heater3 does not work for me, the BL Touch Probe is never deployed. When changing the channel to "!duex.pwm1" manually, it works, but this conflicts the ConfigTool...

I've crashed nozzle hard to the bed once, because of this issue. Definitely needed it fix, before more people crashing their nozzle to bed.

hinotori74, 2020-07-07:
@chrishamm please, reopen the issue.

chrishamm, 2020-07-13:
I'll invert the duex.pwm# channels in the next version.