Duet3D/ConfigTool #46: M143: Max temperature for multiple heaters not generated in config.g

natthapolvanasrivilai, 2020-08-07:
Isn't the M143 command supposed to be generated for every heaters created?

For example,
Heater 0 - Bed Heater
Heater 1 - Hotend Heater
Heater 2 - Chamber Heater

With these 3 heaters configured, shouldn't there be a M143 Hx Sxxx command for every heater?

The config only generate M143 for the first heater, and not the rest.

```
M308 S0 P"bedtemp" Y"thermistor" T100000 B3950         ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bedheat" T0                                  ; create bed heater output on bedheat and map it to sensor 0
M307 H0 B0 S1.00                                       ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                                ; map heated bed to heater 0
M143 H0 S150                                           ; set temperature limit for heater 0 to 150C
M308 S1 P"e1temp" Y"thermistor" T100000 B4725 C7.06e-8 ; configure sensor 1 as thermistor on pin e1temp
M950 H1 C"e1heat" T1                                   ; create nozzle heater output on e1heat and map it to sensor 1
M307 H1 B0 S1.00                                       ; disable bang-bang mode for heater  and set PWM limit
M308 S2 P"spi.cs1" Y"rtd-max31865"                     ; configure sensor 2 as thermocouple via CS pin spi.cs1
M950 H2 C"e0heat" T2                                   ; create chamber heater output on e0heat and map it to sensor 2
M307 H2 B1 S1.00                                       ; enable bang-bang mode for the chamber heater and set PWM limit
M141 H2                                                ; map chamber to heater 2

```

xadox-1st, 2020-09-08:
Same for me. Only M143 H0 S120 was generated.
Temp Limit for E0 ist not set.

chrishamm, 2020-11-16:
Thanks, I'll fix that.

closed 2020-11-16 by chrishamm