Duet3D/ConfigTool #54: Duet 3 Extension Board 3HC OUT Pins incorrect

slideup-benni, 2020-10-06:
Hello,

it looks like the definition of the fan out pins is incorrect, it should be out3 to out8.

https://github.com/Duet3D/ConfigTool/blob/e5161883f73d8702cf22ff5ab4f23fd084a61e66/src/store/ExpansionBoards.js#L37

Please take a look at [https://duet3d.dozuki.com/Wiki/Duet_3_Expansion_3HC_Wiring_Diagram](url)

Can you change this, please.
Thanks, Benjamin

slideup-benni, 2020-10-20:
I found more:
- out3 is missing for Duet 3 Mainboard
- all ioX are missing for z-probe "input pin" for ext3hc
- io2.out, io3.out, io5.out and io0.out are missing for z-probe "pwm control channel" for ext3hc

StephenRC, 2020-10-22:
Also, for the 1LC, you can't select io_0.in for the pwm.

chrishamm, 2020-11-16:
Thanks, I'll add new pin definitions soon.

@slideup-benni io0.out, io2.out, io3.out, and io5.out do not support PWM. See https://duet3d.dozuki.com/Wiki/Duet_3_Expansion_Hardware_Overview#Section_IO

closed 2020-11-16 by chrishamm