Duet3D/ConfigTool #75: [FeatureRequest]:  add Duet 3 Mini 5 as Extension Board in IO Mapping

Avatarsia, 2023-01-22:
### Is your feature request related to a problem? Please describe.

The Mini 5 is completely missing from the list of extension boards, but it is also CANbus capable.

### Describe the solution you propose.

Ex. It would be great if...

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

chrishamm, 2025-01-07:
That's already possible:

[image]

closed 2025-01-07 by chrishamm

Avatarsia, 2025-01-07:
> That's already possible: [...]

was it also 2 years ago ? =D