Duet3D/ConfigTool #78: Download error for reprap fisher on legacy

casesolved-co-uk, 2024-01-29:
Hi,

I get the following error on running the legacy tool for `RepRapPro Fisher`.

I'm not really changing much (just a couple of probe radiuses and the machine name) as I assume (!?) the values are correct.

Any ideas please?

```
TypeError: Cannot read properties of null (reading 'temp_limit')
    at ToolsMixRatio.vue:1:5452
    at a.Ue [as _l] (vue.runtime.esm.js:2002:22)
    at a.Oe (ToolsMixRatio.vue:1:4866)
    at t._render (vue.runtime.esm.js:2655:28)
    at a.r (vue.runtime.esm.js:3831:27)
    at t.get (vue.runtime.esm.js:3407:33)
    at t.run (vue.runtime.esm.js:3483:30)
    at Kr (vue.runtime.esm.js:4077:17)
    at Array.<anonymous> (vue.runtime.esm.js:3114:20)
    at Wn (vue.runtime.esm.js:3036:16)
zn @ vue.runtime.esm.js:3020
vue.runtime.esm.js:3020 Failed to create config.g: TypeError: ejs:308
    306| <%				}
    307| 				if (heater.channel < 100) { -%>
 >> 308| M305 P<%- i %> T<%- heater.thermistor %> B<%- heater.beta + ((heater.c === 0) ? 0 : ' C' + heater.c.toExponential(6)) %> R<%- heater.series %><%- (heater.channel != i) ? ' X' + heater.channel : '' %> ; set thermistor + ADC parameters for heater <%- i %><%- (heater.channel !== i) ? ' and remap it to channel ' + heater.channel : '' %>
    309| <%				} else if (heater.channel >= 200 && heater.channel < 250) { -%>
    310| M305 P<%- i %> X<%- heater.channel %> ; configure PT100 for heater <%- i %>
    311| <%				} else if (heater.channel >= 500 && heater.channel < 550) { -%>

Cannot read properties of undefined (reading 'toExponential')
zn @ vue.runtime.esm.js:3020
```

chrishamm, 2024-01-29:
Using the stock Fisher preset generates a working config.g for me after unchecking "Get latest firmware" and "Get lastest web interface". Can you please share the JSON template from the Finish dialog on last page?

casesolved-co-uk, 2024-01-29:
Thanks!

```json
{
 "board": "duet06",
 "firmware": 1.21,
 "nvram": true,
 "geometry": {
  "type": "delta",
  "delta_radius": 81,
  "rod_length": 160,
  "print_radius": 75,
  "homed_height": 169,
  "z_min": -0.2,
  "max_carriage_travel": 179,
  "mins": [
   0,
   0,
   0
  ],
  "maxes": [
   230,
   210,
   200
  ],
  "low_dive_height": false
 },
 "drives": [
  {
   "direction": 1,
   "microstepping": 16,
   "microstepping_interpolation": false,
   "steps_per_mm": 87.489,
   "instant_dv": 20,
   "max_speed": 250,
   "acceleration": 4000,
   "current": 800,
   "driver": 0,
   "endstop_type": 1,
   "endstop_location": 2,
   "driver_v3": "0.0",
   "endstop_pin": "xstop"
  },
  {
   "direction": 1,
   "microstepping": 16,
   "microstepping_interpolation": false,
   "steps_per_mm": 87.489,
   "instant_dv": 20,
   "max_speed": 250,
   "acceleration": 4000,
   "current": 800,
   "driver": 1,
   "endstop_type": 1,
   "endstop_location": 2,
   "driver_v3": "0.1",
   "endstop_pin": "ystop"
  },
  {
   "direction": 1,
   "microstepping": 16,
   "microstepping_interpolation": false,
   "steps_per_mm": 87.489,
   "instant_dv": 20,
   "max_speed": 250,
   "acceleration": 4000,
   "current": 800,
   "driver": 2,
   "endstop_type": 1,
   "endstop_location": 2,
   "endstop": 2,
   "driver_v3": "0.2",
   "endstop_pin": null
  },
  {
   "direction": 1,
   "microstepping": 16,
   "microstepping_interpolation": false,
   "steps_per_mm": 137,
   "instant_dv": 20,
   "max_speed": 150,
   "acceleration": 4000,
   "current": 1000,
   "driver": 3,
   "driver_v3": "0.3"
  }
 ],
 "idle": {
  "used": true,
  "factor": 30,
  "timeout": 30
 },
 "homing_speed_fast": 30,
 "homing_speed_slow": 6,
 "travel_speed": 200,
 "z_dive_height": 4,
 "slow_homing": false,
 "probe": {
  "type": "switch",
  "trigger_height": -0.1,
  "trigger_value": 500,
  "x_offset": 0,
  "y_offset": 0,
  "speed": 2,
  "deploy": false,
  "points": [
   {
    "x": 0,
    "y": 64.9,
    "z": 0
   },
   {
    "x": 56.21,
    "y": -32.45,
    "z": 0
   },
   {
    "x": -56.21,
    "y": -32.45,
    "z": 0
   },
   {
    "x": 0,
    "y": 0,
    "z": 0
   }
  ],
  "recovery_time": 0.4,
  "pwm_channel": 3,
  "pwm_inverted": true,
  "pwm_pin": null,
  "input_pin": "zprobe.in",
  "modulation_pin": null
 },
 "num_nozzles": 1,
 "bed_is_nozzle": false,
 "bed": {
  "present": true,
  "use_pid": false,
  "heater": 0
 },
 "chamber": {
  "present": false,
  "use_pid": false,
  "heater": 2
 },
 "heaters": [
  {
   "channel": 0,
   "output_pin": null,
   "sensor_pin": null,
   "sensor_type": null,
   "scale_factor": 100
  },
  {
   "scale_factor": 100,
   "series": 4700,
   "thermistor": 100000,
   "beta": 4388,
   "c": 0,
   "channel": 1,
   "sensor": 1,
   "temp_limit": 280,
   "output_pin": "e0heat",
   "sensor_type": "thermistor",
   "sensor_pin": "e0temp"
  }
 ],
 "toolchange_wait_for_temperatures": true,
 "generate_t_code": false,
 "tools": [
  {
   "number": 0,
   "extruders": [
    0
   ],
   "heaters": [
    1
   ],
   "x_offset": 0,
   "y_offset": 0,
   "z_offset": 0,
   "mix_ratio": [

   ],
   "name": "Nozzle",
   "fans": [
    0
   ]
  }
 ],
 "peripheral_points": 3,
 "halfway_points": 0,
 "calibration_factors": 3,
 "probe_radius": 65,
 "mesh": {
  "radius": 65,
  "spacing": 20,
  "x_min": 15,
  "x_max": 215,
  "y_min": 15,
  "y_max": 195
 },
 "home_first": false,
 "orthogonal": {
  "compensation": false,
  "height": 50,
  "deviations": [
   0,
   0,
   0
  ]
 },
 "network": {
  "enabled": true,
  "name": "MenShed",
  "password": "MenShed",
  "mac_address": "BE:EF:DE:AD:FE:ED",
  "dhcp": true,
  "ssid": "",
  "ssid_password": "",
  "ip": "192.168.1.20",
  "netmask": "255.255.255.0",
  "gateway": "192.168.1.254",
  "protocols": {
   "http": true,
   "ftp": false,
   "telnet": false
  }
 },
 "fans": [
  {
   "value": 0,
   "inverted": false,
   "frequency": 500,
   "thermostatic": false,
   "name": "",
   "heaters": [

   ],
   "trigger_temperature": 45,
   "output_pin": "fan0"
  }
 ],
 "custom_settings": "M301 H1 P12 I0.4 D80 W180 B300 ; Set nozzle heater PID parameters",
 "expansion_boards": [

 ],
 "standalone": true,
 "auto_save": {
  "enabled": false,
  "save_threshold": 10,
  "resume_threshold": 11,
  "gcodes_to_run": "M913 X0 Y0 G91 M83 G1 Z3 E-5 F1000"
 },
 "display": {
  "type": 0,
  "encoder_steps": 4,
  "spi_frequency": 2000000,
  "menus": [
   {
    "name": "main",
    "value": ""
   }
  ],
  "images": [

  ]
 },
 "panelDue": false,
 "compensation_x_offset": 15,
 "compensation_y_offset": 15
}
```

casesolved-co-uk, 2024-01-29:
> Using the stock Fisher preset generates a working config.g for me after unchecking "Get latest firmware" and "Get lastest web interface". Can you please share the JSON template from the Finish dialog on last page?

The problem appears to be when I enable `heated bed`

chrishamm, 2024-01-29:
Thanks, fixed in legacy configtool version 3.4.1.

closed 2024-01-29 by chrishamm

casesolved-co-uk, 2024-01-29:
@chrishamm you also need to have an option for geared vs direct drive extruders.
Your config appears to be for direct drive.

M92 X87.49 Y87.49 Z87.49 E310.00            ; set steps per mm. E137-E144 for direct drive, E310 for geared drive extruder