Duet3D/ConfigTool #79: Multiple Heater for 1 tool

Gradode, 2024-03-27:
I've noticed a few bugs.

For RRF version 3.5.

Firstly, when we add more than 2 heaters for a single tool, the + button is inaccessible, I don't know if I'm the only one with this problem. I declared 4 heaters, 3 for a head and 1 for the bed

Secondly, I don't understand why when we configure several motors per axis, we have to add several endstops otherwise we can't save our configuration. I've managed to get around the problem by declaring other io pins, but in the config file these "false io pins" don't appear.

[image]

chrishamm, 2024-04-11:
Thanks for reporting the problem affecting endstops, it's going to be fixed in 3.5.0-rc.4. I guess you are using a Mini 5+. If you are, the limit of heaters per tool is 2 in the firmware so the configtool has the same limitation.

closed 2024-04-11 by chrishamm

Gradode, 2024-04-11:
> Thanks for reporting the problem affecting endstops, it's going to be fixed in 3.5.0-rc.4. I guess you are using a Mini 5+. If you are, the limit of heaters per tool is 2 in the firmware so the configtool has the same limitation.

No, I use and select the MB duet3 6HC.
But I export JSON configuration and I modify this to add my heater 3.

chrishamm, 2024-04-11:
Ah sorry, my bad, that should be fixed now. The limits were not correctly updated when the board type was changed.